COVESA/dlt-viewer #654: Optimize file indexing

RuiMarioCosta, 2025-04-26:
dtl-viewer is slow when loading DLT files taking some time to load a file with just 500mb and becomes extremely slow files in the gigabytes. There are several parts that contribute to this but the main method is QDltFile::getMsg which is called from DltFileIndexer::indexFilter.

My plan is to first stablish a baseline to understand if the performance is actually improving or not and to do that I was thinking of using [google's benchmark](https://github.com/google/benchmark/tree/main) library which is fairly easy to use and well documented. The idea in this step is to create bechmarks for the biggest hotspots of the code. After that the optimization can start.

I already have some small benchmarks but nothing special but before putting more effort would this be a good idea?

alexmucde, 2025-04-28:
@RuiMarioCosta Good idea to find the bottlenecks and possibilities for optimization

alexmucde, 2025-07-02:
@RuiMarioCosta QCache is already used to cache the messages in memory. The Benefit is, that the messages are already parsed. The size of the cache can be set here:

[image]

But this cache does not really help when having big files, only the crolling is getting smoother.

Another improvement could be when the Filter index runs, already showing the results to the user, when the indexer is still running, so that the user has not to wait for the end. Thats the same like the Search function is already doing.

sudomgamal, 2025-07-06:
I also see multiple operations done inside the GUI event loop (in the GUI related slots). This can impact the application responsiveness since the operations can take a lot of time specially in the cases mentioned by @RuiMarioCosta

Besides the performance improvements, we can use a worker queue to handle these operations outside the event loop. But maybe this can be tracked on another ticket.

RuiMarioCosta, 2025-07-19:
@sudomgamal, maybe that can also be optimized, but from the profiling results, the program spends most of the time in the `getMsg` and `setMsg`. I'm writing a small POC to see if the ideas that I have can improve my concern is how to later incorporate them here.

I would really like this to be improved because just the other day, I had to open a ~3Gb file, and I think it took almost 10 minutes.