COVESA/dlt-viewer #688: Export with RegEx Replace in DLT Filter does not work in dlt-commander

alexmucde, 2025-07-09:
If you use the export functionality by command line using a filte rconfiguration with Regex Replace function the results are different for dlt-viewer and dlt-commander

dlt-commander INFILE -d -c OUTFILE.commander.dlt  FILTER
dlt-viewer          INFILE -s -t -d -c OUTFILE.viewer.dlt   FILTER 

For dlt-commander the RegexReplace does not work at all.

alexmucde, 2025-07-11:
@vifactor Do you find some time to look into this issue in the next days? I would like to have it fixed for the 2.2.8.0 release.

alexmucde, 2025-07-11:
Ok i will have a look by myself

vifactor, 2025-07-13:
@alexmucde please check this: https://github.com/COVESA/dlt-viewer/pull/690 there were indeed couple of issues after my changes. The only way to avoid that is to write a nice set of tests for the `commander`, which can be as simple as running the tool using python script for various set of parameters and inspecting output.

closed 2025-07-17 by vifactor